quartiq/booster #290: Booster not able to properly boot

ryan-summers, 2023-03-31:
The booster that I have in my office spuriously started to fail to properly boot recently.

Observations:
* The fan self test occurs at a normal period 
* DFU to version 0.4. worked and properly booted
* Recent firmware versions on `main` do not work going back a few months

ryan-summers, 2023-03-31:
Commenting out the watchdog and running via `cargo run` results in the app never finishing initialization. Backtrace as follows:
```
    Finished dev [optimized + debuginfo] target(s) in 41.81s
     Running `probe-run --chip STM32F407ZGTx 'target\thumbv7em-none-eabihf\debug\booster'`
(HOST) INFO  flashing program (462.34 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
INFO - Starting initialization
────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: <synopsys_usb_otg::bus::UsbBus<USB> as usb_device::bus::UsbBus>::enable::{{closure}}
        at C:\Users\rsummers\.cargo\registry\src\github.com-1ecc6299db9ec823\synopsys-usb-otg-0.2.3\src\bus.rs:262:19
   1: cortex_m::interrupt::free
        at C:\Users\rsummers\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-0.7.7\src\interrupt.rs:64:13
   2: <synopsys_usb_otg::bus::UsbBus<USB> as usb_device::bus::UsbBus>::enable
        at C:\Users\rsummers\.cargo\registry\src\github.com-1ecc6299db9ec823\synopsys-usb-otg-0.2.3\src\bus.rs:256:9
   3: core::cell::Cell<T>::get
        at /rustc/9eb3afe9ebe9c7d2b84b71002d44f4a0edac95e0/library/core/src/cell.rs:452:18
   4: <core::cell::BorrowRefMut as core::ops::drop::Drop>::drop
        at /rustc/9eb3afe9ebe9c7d2b84b71002d44f4a0edac95e0/library/core/src/cell.rs:1670:34
   5: core::ptr::drop_in_place<core::cell::BorrowRefMut>
        at /rustc/9eb3afe9ebe9c7d2b84b71002d44f4a0edac95e0/library/core/src/ptr/mod.rs:490:1
   6: core::ptr::drop_in_place<core::cell::RefMut<synopsys_usb_otg::bus::UsbBus<stm32f4xx_hal::otg_fs::USB>>>
        at /rustc/9eb3afe9ebe9c7d2b84b71002d44f4a0edac95e0/library/core/src/ptr/mod.rs:490:1
   7: usb_device::bus::UsbBusAllocator<B>::freeze
        at C:\Users\rsummers\.cargo\registry\src\github.com-1ecc6299db9ec823\usb-device-0.2.9\src\bus.rs:174:39
   8: usb_device::device::UsbDevice<B>::build
        at C:\Users\rsummers\.cargo\registry\src\github.com-1ecc6299db9ec823\usb-device-0.2.9\src\device.rs:93:19
   9: booster::hardware::setup::setup
        at src\hardware\setup.rs:425:9
  10: booster::app::init
        at src\main.rs:80:27
  11: booster::app::rtic_ext::main::{{closure}}
        at src\main.rs:55:1
  12: booster::app::rtic_ext::main::__rtic_init_resources
        at src\main.rs:55:1
  13: main
  14: ResetTrampoline
        at C:\Users\rsummers\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rt-0.6.15\src\lib.rs:547:26
  15: Reset
        at C:\Users\rsummers\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-rt-0.6.15\src\lib.rs:550:13
(HOST) INFO  device halted by user
error: process didn't exit successfully: `probe-run --chip STM32F407ZGTx 'target\thumbv7em-none-eabihf\debug\booster'` (exit code: 2)
```

ryan-summers, 2023-03-31:
That traceback links back to a spin-wait in the USB bus driver:
[image]

It looks like the USB driver isn't detecting the AHB idle condition as expected

closed 2023-03-31 by ryan-summers

gkasprow, 2023-03-31:
Is that anything HW-related?
Did you check the 3.3V power rail?

ryan-summers, 2023-03-31:
No, I highly doubt this is hardware related. It seemed like a race condition in firmware causing some strange state of the clocks, and updating the various booster dependencies resolved things.

PR #291 (2023-03-31): Updating HAL and other dependencies
This PR updates the HAL to remove usage of a patch and keep us using released firmware.

This PR fixes #290 by (presumably) fixing issues with the USB driver or HAL respectively.

I ran a self-test to confirm normal operation.

```
-> Conducting self-test on channel 0
Setting output interlock threshold to 30 dB
Commanding channel 0 into Enabled
Setting output interlock threshold to -5 dB, verifying interlock trips
Commanding channel 0 off
Channel 0: PASS

-> Conducting self-test on channel 1
Setting output interlock threshold to 30 dB
Commanding channel 1 in [... cut]