quartiq/booster #380: Enable settings modification via USB

ryan-summers, 2024-03-25:
We should allow users to configure the device runtime settings via the USB interface similar to Stabilizer. This would enable the tool to be used without MQTT for initial setup.

ryan-summers, 2024-03-25:
Although the channel tuning algorithms etc. still require MQTT, so this may have less use than Stabilizer.

CC @jordens: Do you think this is worthwhile doing for Booster, or should we leave as-is?

jordens, 2024-04-23:
TODO: port over to stabilizer serial settings

jordens, 2024-06-13:
#400

ryan-summers, 2024-06-14:
@jordens This issue was actually asking about runtime settings modifications via USB, like modifying interlock settings etc. similar to how we support this on Stabilizer.

jordens, 2024-06-14:
Doesn't that come for free now with serial settings?

ryan-summers, 2024-06-14:
We still need to restructure the application-level logic a bit. Currently, just the `BoosterMainBoard` settings are exposed via USB, so the operator can only modify basic things like `ip`, `broker`, etc.

This was asking if we should expose channel settings etc. over USB as well. It sounds like yes.

jordens, 2024-06-14:
Ack. Yes. Let's do it like on stabilizer, thermostat-eem, fls.

PR #403 (2024-06-20): Refactoring settings to enable USB control of runtime-settings
This PR fixes #380 by allowing RF channel settings to be configured via the USB port.

TODO:
- [x] Currently, `serial-settings` will always say that you need to reboot to apply settings. This isn't true for runtime settings.
- [ ] ~Save RF channel transforms into the RF channel EEPROM, not device flash~
    * Deferred to https://github.com/quartiq/booster/issues/404
- [x] Make `clear` reset the RF channel configurations
closed 2024-06-20 by jordens