quartiq/booster #404: Support saving settings to RF channel EEPROMs

ryan-summers, 2024-06-18:
It was noted that certain settings, such as the power transformations, make the most sense existing on the physical RF channel memory. This would ensure that the transforms would move if the RF channel was installed in a separate device.

We should update the settings `save` feature to save the RF transforms to RF channel EEPROMs instead of the main board flash.

ryan-summers, 2024-06-20:
@jordens Upon reviewing the code, it looks like the RF channel transforms are largely dependent on the input signal frequency, so the linear transforms are likely heavily dependent on the input signal as opposed to the hardware characteristics. Given this condition, it seems like the transforms may make more sense being stored in main-board flash, since I would expect that the input signals would likely be changing around if Booster's internal RF channels were reorganized.

jordens, 2024-06-20:
Slope will likely depend on PVT as much as the offset.
But on balance I agree that the benefit of making the transforms (physically) portable with the modules is small and the complexity is high. Let's just drop the per-channel configuration in EEPROM and move everything in to the CPU flash as in the other projects. Also make the `save` functionality match stabilizer et al.

PR #407 (2024-06-20): Feature/clear support
Adds `clear <item>` support. This also fixes #404 by removing the TODO comment in the source code.
closed 2024-06-20 by jordens